PrusaOwners/Marlin #10: Add more temperature profiles

matthew-humphrey, 2019-02-18:
Marlin supports two temperature profiles, PLA and ABS, that appear in the LCD menu. We should investigate how difficult it would be to add more, and what the impact might be on re-integration with Marlin. If not do onerous, we should add ones for PETG and maybe other filaments.

CapitaineCrunche, 2019-02-20:
For temperature profiles is easy to add more but it will impact Marlin because we need to add lines in many libraries. i can do it in my fork of Marlin. PLA, ABS and PETG and what else?

matthew-humphrey, 2019-02-20:
Can you elaborate on where this code is in Marlin? I want to look and see if there's a way to make it more modular.

CapitaineCrunche, 2019-02-21:
line 1895 in configuration_store.cpp, line 1267 in configuration.h, line 115 in language_en.h(and other language) and the rest in ultralcd.cpp.

CapitaineCrunche, 2019-02-21:
maybe this will be more simple way to do it:
MENU_ITEM(gcode,MSG_PREHEAT_1, PSTR(
"M104 S215\n"
"M140 S80"
));