PrusaOwners/Marlin #13: MK3S filament sensor

stahlfabrik, 2019-02-18:
Evaluate how to add support for MK3s filament sensor. 
Maybe Marlin already supports it. Pins?!

matthew-humphrey, 2019-02-18:
It definitely supports this, we just need to configure it.

matthew-humphrey, 2019-02-19:
In fact, I suspect no changes will be necessary over the current sensor. I think it plugs into the same place on the board, and Prusa designed the hardware so that the switch interface (out / not out) to be compatible with their firmware.  But we need to verify this. 

### Detail ###
Currently in configuration the setting is in Configuration.h. We need to define somewhere:

`#define FIL_RUNOUT_PIN ?

    #if ENABLED(FILAMENT_RUNOUT_SENSOR)
      #define NUM_RUNOUT_SENSORS   1     // Number of sensors, up to one per extruder. Define a FIL_RUNOUT#_PIN for each.
      #define FIL_RUNOUT_INVERTING false // set to true to invert the logic of the sensor.
      #define FIL_RUNOUT_PULLUP          // Use internal pullup for filament runout pins.
      #define FILAMENT_RUNOUT_SCRIPT "M600"
    #endif

stahlfabrik, 2019-02-19:
This might help us:

https://manual.prusa3d.com/Guide/4.+Electronics+assembly/1061?lang=en#s18777

matthew-humphrey, 2019-02-19:
Actually this was helpful:
https://manual.prusa3d.com/Guide/3.+E-axis+assembly/1060?lang=en

I checked this against the Einsy schematic. They are connecting the sensor to connector P3 on the Einsy, with +5 from pin 2, GND from pin 4, and the sensor output to pin 10. Pin 10 is connected to PK0 I/O pin on microcontroller. Checking fastio_1280.h, this is logical pin 62.

stahlfabrik, 2019-02-19:
You are right!

#define IR_SENSOR_PIN 62 //idler sensor @PK0 (digital pin 62/A8)

Source:
https://github.com/prusa3d/Prusa-Firmware/blob/MK3S/Firmware/pins_Einsy_1_0.h

matthew-humphrey, 2019-02-19:
Also, I confirmed in the Prusa firmware that a logic HIGH input on the sensor pin indicates the filament is NOT present (i.e. has run out). Here is some relevant code from fsensor.cpp

		if ((digitalRead(IR_SENSOR_PIN) == 1) && CHECK_FSENSOR && fsensor_enabled && ir_sensor_detected)
		{
			fsensor_stop_and_save_print();
			printf_P(PSTR("fsensor_update - M600\n"));
			eeprom_update_byte((uint8_t*)EEPROM_FERROR_COUNT, eeprom_read_byte((uint8_t*)EEPROM_FERROR_COUNT) + 1);
			eeprom_update_word((uint16_t*)EEPROM_FERROR_COUNT_TOT, eeprom_read_word((uint16_t*)EEPROM_FERROR_COUNT_TOT) + 1);
			enquecommand_front_P(PSTR("FSENSOR_RECOVER"));
			enquecommand_front_P((PSTR("M600")));
		}

matthew-humphrey, 2019-02-19:
So looks like we have an issue. As mentioned above, the signal is active high. The pin has a pull-up resistor configured, which want so that if it's connected, the input is not floating. However, when that happens, it will trigger filament sensor runout. This behavior should be the same on Prusa firmware. However, unlike Prusa, Marlin has no way to disable sensor other than turn it off in Configuration.h and rebuild.