PrusaOwners/Marlin #34: Bootloop when starting with SD Card In

pnoll1, 2019-02-26:
Printer will not boot with SD card inserted.

Workaround is to pull out card and reinsert after boot or format card and restart.

SD card is 8GB GoodRam that came with printer, has preloaded Prusa files and a few others.

stahlfabrik, 2019-02-26:
Maybe that bug is fixed in main Marlin already? Someone should check with Marlin 2.x code:-)

I also experienced this. I could imagine it has to do with certain chars in filenames or something along those lines. Or hidden files.

matthew-humphrey, 2019-02-26:
I've been booting worth an SD card inserted for weeks.

stahlfabrik, 2019-02-26:
My problem is also gone since I formatted the sd again. Before that there various old files that prusa firmware had no problem with but marlin crashed during boot

matthew-humphrey, 2019-02-26:
I did notice an SD card related big fix in the bugfix-2.0 branch, but I didn't look at the details

pnoll1, 2019-02-26:
could add a blurb to format the sd in the readme and close this.

matthew-humphrey, 2019-02-26:
Would be good to capture a way to repro so if we come back later to fix we can recreate. Do we know what file or other condition triggers it? Am wondering if it's a file length issue, or possibly the number of files (alpha sort is limited to 100 files in our config)

pnoll1, 2019-02-26:
Files on card:
 - MK3_PLA_3DBenchy_150um_2H.gcode
 - MK3_PLA_3DHubs_Marvin_100um_1H_7M.gcode
 - MK3_PLA_Adalinda_200um_7H_45M.gcode
 - MK3_PLA_Batman_200um_18M.gcode
 - MK3_PLA_Buddy_150um_2H.gcode
 - MK3_PLA_Castle_100um_13H_20M.gcode
 - MK3_PLA_Gear_Bearing_150um_2H_20M.gcode
 - MK3_PLA_Nefertiti_150um_7H_50M.gcode
 - MK3_PLA_Prusa_200um_19M.gcode
 - MK3_PLA_PrusaBeerOpener_ColorPrint_200um_46M.gcode
 - MK3_PLA_Treefrog_150um_variable_1H_17M.gcode
 - MK3_PLA_Treefrog_50um_3H_40M.gcode
 - MK3_PLA_Triceratops_Skull_150um_5H_10M.gcode
 - MK3_PLA_Vase_200um_6H_25M.gcode
 - MK3_PLA_Whistle_200um_30M.gcode
 - MK3_PRUSA_dual_color.gcode
 - battery holder dual v11 with modifiers3 MK3 ret 0.08 LA.14up.gcode
- jigsaw tool holder.gcode
 - 'System Volume Information'
   - IndexerVolumeGuid  
   - WPSettings.dat

4 files around 20MB, 2 more at 10MB, rest are smaller.

matthew-humphrey, 2019-02-27:
Interesting! I created a little PowerShell script to create empty files for all the names you have up there, except the System Volume Information folder, and I can easily reproduce the error. I'm attaching the script in case anyone else is interested. You need to run it from PowerShell with the root of the SD card as the current directory.

[create_files.zip](https://github.com/PrusaOwners/Marlin/files/2908446/create_files.zip)

matthew-humphrey, 2019-02-27:
yeah, so it looks like forking off 1.1.9 was a really bad decision. There are fixes in bugfix-1.1.x that we want, including this gem:

https://github.com/MarlinFirmware/Marlin/commit/a7398a0459973abd0056f0c2051d1fae8375e67c#diff-d609751f5615a9d38385992f80b590bc

pnoll1, 2019-03-01:
Corrected file list:

    * MK3_PLA_3DBenchy_150um_2H.gcode

    * MK3_PLA_3DHubs_Marvin_100um_1H_7M.gcode

    * MK3_PLA_Adalinda_200um_7H_45M.gcode

    * MK3_PLA_Batman_200um_18M.gcode

    * MK3_PLA_Buddy_150um_2H.gcode

    * MK3_PLA_Castle_100um_13H_20M.gcode

    * MK3_PLA_Gear_Bearing_150um_2H_20M.gcode

    * MK3_PLA_Nefertiti_150um_7H_50M.gcode

    * MK3_PLA_Prusa_200um_19M.gcode

    * MK3_PLA_PrusaBeerOpener_ColorPrint_200um_46M.gcode

    * MK3_PLA_Treefrog_150um_variable_1H_17M.gcode

    * MK3_PLA_Treefrog_50um_3H_40M.gcode

    * MK3_PLA_Triceratops_Skull_150um_5H_10M.gcode

    * MK3_PLA_Vase_200um_6H_25M.gcode

    * MK3_PLA_Whistle_200um_30M.gcode

    * MK3_PRUSA_dual_color.gcode

    * batter~1.gco

    * jigsaw tool holder.gcode

    * 'System Volume Information'
      
      * IndexerVolumeGuid
      * WPSettings.dat

Deleting the longest filename (MK3_PLA_PrusaBeerOpener_ColorPrint_200um_46M.gcode) lets the printer boot

stahlfabrik, 2019-03-04:
There are also some SD config options that might be useful.
e.g:
  // This allows hosts to request long names for files and folders with M33
  //#define LONG_FILENAME_HOST_SUPPORT